stlink-org/stlink #1435: [0483:3748 STMicroelectronics ST-LINK/V2]: Not recognized in st-gui

mesb1, 2024-10-10:
**Thank you for giving feedback to the stlink project.**

---

**NOTE: In order to offer sufficient and the best possible support, please read /CONTRIBUTING.md and follow the given instructions _before_ submitting a ticket.**

**Bug reports and/or feature requests will be deleted, if they violate our contribution guidelines and if no issue-template is used!** Thank you for your support.

---

- [x] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers to isolate and target your respective issue, please take some time to select the check boxes below and fill out each of the following items appropriate to your specific problem.

- [x] Programmer/board type: [ID 0483:3748 STMicroelectronics ST-LINK/V2] (e.g STLINK /V1, /V2, /V2-onboard, /V2-clone, /V3)
- [x] Operating system an version: [Linux Mageia 9] (e.g Linux, Windows)
- [x] **stlink tools version** and/or git commit hash: [1.8.0] (e.g v1.6.1/git-d0416149)
- [x] stlink commandline tool name: [st-info] (e.g `st-info`, `st-flash`, `st-trace`, `st-util`)
- [x] Target chip (and board, if applicable): [Not sure how to get it] (e.g STM32F103C8T6 (NUCLEO-F103RB))

lsusb -v 

Bus 002 Device 003: ID 0483:3748 STMicroelectronics ST-LINK/V2
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0483 STMicroelectronics
  idProduct          0x3748 ST-LINK/V2
  bcdDevice            1.00
  iManufacturer           1 STMicroelectronics
  iProduct                2 STM32 STLink
  iSerial                 3 W
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0027
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              4 ST-Link Debug
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)
Further we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

Commandline output:

```
OUTPUT/ERROR of the commandline tool(s)
st-info --probe
Found 0 stlink programmers
```

Expected/description:

I have some small usb stlink/v2 device that looks like a small usb flash in a green color.
I was hope i can use it with this tools to flash my stm based clock under linux.
Unfortunately gui says chip unknown, and no st-info probe is shows it connected.

Is there anythin i can help to make it work?

Nightwalker-87, 2024-10-19:
From your description this sounds like one of these cheap clone/(fake) stlink/V2 programmer to me.
They may work, they may work partially only or the may even not work at all.

So far I don't see a bug here. The observed behaviour may be one of those risks when using these st-link clone programmers. If one can not verify that the assembled chip is an original part, everything can happen...
Please use original hardware or at least make sure that your cloned device has a original chip manufactured by ST-Microelectronics by replacing the chip itself.

Further details on this issue can be found in a separate section in our documentation at `doc/tutorial.md` and also in the following resolved issues: #493, #840, #1012, #1206, #1308 and in the same context also #1322.

Nightwalker-87, 2024-11-09:
Unfortunately we are not able to verify or further investigate this due to missing feedback and contribution.
As appropriate advice has been given, I am closing this ticket now as resolved.

closed 2024-11-09 by Nightwalker-87