stlink-org/stlink #1436: [feature] STM32U073 support

kevlar700, 2024-10-16:
Is there any kind of expected support timeline for STM32U073 in either stlink or openocd. I use Ada and need to connect gdb via something. Thank You.

- [ ] Programmer/board type: STLINK V3 MINIE
- [ ] Operating system and version: Linux Debian 12 Bookworm (Devuan actually but the same thing really)
- [ ] STM32U073

Nightwalker-87, 2024-10-17:
No, sorry - not in the foreseeable future, unless any volunteer is going to take action here.

Ant-ON, 2024-10-17:
@kevlar700 U073 has similar flash peripheral to C0xx chips . You can create new config for U073 based on C0xx config.

Nightwalker-87, 2024-11-30:
A first attempt can be found here: https://github.com/markxoe/stlink/commit/82741108113cd3af222a8ec4ed3ba316eb33cbfc.
However the device type still needs to be defined.
I can't say though, if additional changes are necessary for these devices.

commit 11e357ae (2025-01-08): General project update

- Updated version_support
- Updated tutorial
- Updated CHANGELOG
- Updated list of contributors
- Fixed parsing of unit prefixes (Closes #1445)
- Updated list of supported devices
- Added support for STM32C051/71/91 chips
- Added support for STM32U031/73/83 chips (Closes #1436)
closed 2025-01-08 by Nightwalker-87