stlink-org/stlink #1444: [feature] Release v1.8.1

jpteb, 2024-12-27:
Hi stlink team,

First of all, thank you for these excellent tools!

I don’t have a bug report, hopefully just a small request: Would it be possible to release version `1.8.1`? With the switch to GCC 14 coming up, the current release (`v1.8.0`) fails to build due to the `-Wcalloc-transposed-args` warning. This issue has already been addressed in [this commit](https://github.com/stlink-org/stlink/commit/6a6718b3342b6c5e282a4e33325b9f97908a0692) and [this issue](https://github.com/stlink-org/stlink/issues/1365), but the fix hasn’t yet been included in a formal release.

Thank you again for your hard work! Please let me know if there’s any way I can assist.

Nightwalker-87, 2024-12-30:
Thank you for the positive feedback. :+1:
We are working on this release, but are not ready yet.
With more participants (contributors for coding and testing) on the remaining open issues we may proceed more quickly...

jpteb, 2024-12-30:
Ah, I didn’t notice the milestone. There’s no urgency on my end, so no pressure at all.

I’ll see if I can help with the open issues, but I can’t make any promises since I don’t own an H563ZI and don’t use Windows.

Nightwalker-87, 2025-01-06:
@jpteb Having addressed this issue in commit 6a6718b3342b6c5e282a4e33325b9f97908a0692, I'll close this as resolved.
The new release will be published as soon as the milestone is completed.
Depending on the rate of contribution and additional help hopefully this is the case somewhen during Q1/2025.
I'll continue to maintain relevant issues with additional fixes and updates frequently.

closed 2025-01-06 by Nightwalker-87