stlink-org/stlink #1445: [St-util Bug]: $--freq parameter case sensitive in 1.8.0 but not on previous versions

AgustinOrdonez, 2025-01-04:
**Thank you for giving feedback to the stlink project.**

---

**NOTE: In order to offer sufficient and the best possible support, please read /CONTRIBUTING.md and follow the given instructions _before_ submitting a ticket.**

**Bug reports and/or feature requests will be deleted, if they violate our contribution guidelines and if no issue-template is used!** Thank you for your support.

---

- [x] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers to isolate and target your respective issue, please take some time to select the check boxes below and fill out each of the following items appropriate to your specific problem.

- [x] Programmer/board type: [ALL] (e.g STLINK /V1, /V2, /V2-onboard, /V2-clone, /V3)
- [x] Operating system an version: [Windows 10] (e.g Linux, Windows)
- [x] **stlink tools version** and/or git commit hash: [1.8.0] (e.g v1.6.1/git-d0416149)
- [x] stlink commandline tool name: [st-util/st-flash/st-info] (e.g `st-info`, `st-flash`, `st-trace`, `st-util`)
- [x] Target chip (and board, if applicable): [ALL] (e.g STM32F103C8T6 (NUCLEO-F103RB))

Further we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

Commandline output:

```
C:\Program Files (x86)\stlink\bin>st-util -p 44622 --freq=1200K -v
Can't parse a frequency: 1200K
```

Expected/description:

`It starts with 1200kHz frequency, in 1.7.0 this worked fine(upper or lower case "k") . On st-info, st-flash and st-util's help it is lowercase on 1.8 and uppercase on 1.7. This is an issue mainly for automated systems, which used capital K for 1.7 and no longer works on 1.8`

maxim-banaev, 2025-01-06:
Caused by 226e5337610b0d1ab12fd43876f66dc7be2dc626 cc @Nightwalker-87

Nightwalker-87, 2025-01-06:
@maxim-banaev No, this was only white space change - thus further research is needed.

maxim-banaev, 2025-01-06:
@Nightwalker-87 sorry. I was confused by the similar commit title. Could you please take a look at the a06d135.
There is a [change](https://github.com/stlink-org/stlink/commit/a06d13552da5884bd994fddf2655db243e3a6a48#diff-357481b352786e0156a812d2756c5a412f073791d97c020c1406696c28b5b96bL24) that could affect the behaviour.

Nightwalker-87, 2025-01-07:
Should be easy to fix. I'll address it soon.
However, besides backward compatibility, users are strongly encouraged to use established SI-based [unit prefixes](https://en.wikipedia.org/wiki/Unit_prefix). Thus we'll take that into account regarding the output of the help commands.

Nightwalker-87, 2025-01-07:
To me this is how it should actually look like:

```
int32_t arg_parse_freq(const char *str) {
    int32_t value = -1;
    if (str != NULL) {
            char* tail = NULL;
        value = strtol(str, &tail, 10);
        if (tail != NULL) {
            if (tail[0] == 'M' && tail[1] == '\0') {
                value = value * 1024; // Prefix: Mega
            } else if ((tail[0] == 'k' || tail[0] == 'K') && tail[1] == '\0') {
                // value              // Prefix: kilo
            } else if (((tail[0] != 'M' && tail[0] != 'k' && tail[0] != 'K') && tail[1] != '\0') || tail[0] != '\0') {
                value = -1;           // error
            }
        }
    }
    return value;                     // frequency in kHz
}
```

Nightwalker-87, 2025-01-07:
To me this is how it should actually look like:

```
int32_t arg_parse_freq(const char *str) {
    int32_t value = -1;
    if (str != NULL) {
            char* tail = NULL;
        value = strtol(str, &tail, 10);
        if (tail != NULL) {
            if (tail[0] == 'M' && tail[1] == '\0') {
                value = value * 1000 * 1000; // prefix: Mega
            } else if ((tail[0] == 'k' || tail[0] == 'K') && tail[1] == '\0') {
                value = value * 1000;        // prefix: kilo
            } else if (tail[0] == '\0') {
                // value                     // no prefix
            } else {
                value = -1;                  // error
            }
        }
    }
    return value;                            // frequency in Hz
}
```

a-michelis, 2025-01-07:
LGTM! I suppose "m" is missing to avoid "micro-mega" confusion.

As for the readability/ease-of-maintenance, I'd suggest the following:

```c
int32_t arg_parse_freq(const char *str) {
    if (str == NULL) {
        return -1;  // Faulty Input
    }
    
    char* tail = NULL;
    int32_t value = strtol(str, &tail, 10);
    if (tail == NULL) {
        return -1; // Faulty integer conversion
    }

    if (tail == '\0') {
        return value; // no prefix -> Hz
    }
    if (tail[0] == 'M' && tail[1] == '\0') {
        return value * 1000 * 1000; // M prefix ->MHz = 1000_000Hz
    } 
    if ((tail[0] == 'k' || tail[0] == 'K') && tail[1] == '\0') {
        return value * 1000;        // k/K prefix -> KHz = 1000Hz
    }
    return -1; // Invalid Prefix
}
```

I find it easier to read/debug, but that's may just be me and my quirky styling prefferences 😝

Nightwalker-87, 2025-01-07:
Thanks @a-michelis. :+1:
I'll consider this input and expect to complete this by tomorrow evening along with other prepared contributions.

a-michelis, 2025-01-08:
**@Nightwalker-87 heads up:**

I've made a typo on the code I commented yesterday!
[image]

should be `*tail == '\0'`, not `tail == '\0'`

Nightwalker-87, 2025-01-08:
@a-michelis: Thanks for the note, should have seen that as well.

Unfortunately my Github authentication is currently messed up. Thus this commit should have actually gone to the official _stlink-org_ account... Now I find a personal project copy forked from another user fork of the main repository. There were times where things like this simply worked without any need for extraordinary further studies... :vomiting_face: Need to fix this.

closed 2025-01-08 by Nightwalker-87

Nightwalker-87, 2025-01-09:
Finally fixed in commit 7900006619ce22cb900151120f9aeebe022cd3ec.

commit 11e357ae (2025-01-08): General project update

- Updated version_support
- Updated tutorial
- Updated CHANGELOG
- Updated list of contributors
- Fixed parsing of unit prefixes (Closes #1445)
- Updated list of supported devices
- Added support for STM32C051/71/91 chips
- Added support for STM32U031/73/83 chips (Closes #1436)