stlink-org/stlink #1446: [GH_ACTIONS] [BUG] All tests are currently performed on GCC 9.4.0

a-michelis, 2025-01-07:
As I mentioned on https://github.com/stlink-org/stlink/pull/1440#issuecomment-2561168962 , I started looking at the C/C++ Workflow in order to attempt and unify the approach. While doing so, I discovered that all of the CLANG tests were NOT using clang, and used GCC instead.

The reason behind this is that all ubuntu workers on GH are equipped with both gcc-10 and clang-12. Installation of these two was doing essentially nothing, and since GCC was the preffered by cmake (unless specifically requested by setting env variables), all clang tests are running with gcc.

**Solution:**
For clang tests, use `env` section in `.yaml` to define `CC` and `CXX`  variables

```yaml
  job_linux_22_04_64_clang:
    name: ubuntu-22.04 clang
    runs-on: ubuntu-22.04
    env:
      CC: clang-12
      CXX: clang++-12
```

Continuing further, I discovered that, when using CLang in release, CMake appends some integer conversion warning-as-error flags, which are not possible to be removed unless we interfere directly with the cmake files (possibly). That said, the only issue here was a simple implicit casting on libstlink's `helper.cpp` which, when explicitly casted, stopped producing the error

`a/src/stlink-lib/helper.h:29`:
```c
value = strtol(str, &tail, 10);
```

`b/src/stlink-lib/helper.h:29`:
```c
value = (int32_t)strtol(str, &tail, 10);
```

After that I discovered another major issue-duo:
In addition to CLang tests, GCC tests were not using the correct GCC version as well - again solved by defining CC and CXX env variables. BUT, this time it was obvious that `M32` flags were not being registered as well, which led me to add the _workflow-intended_ way (using `env` section within the step). This led to another major issue which is: The compilation fails when using `m32` in both gcc-10 and clang-12, and I still can't figure out why, except the fact that it is related to the linking process.

Ideas? Pointers? Suggestions? All welcome!

A.M.

Nightwalker-87, 2025-01-07:
Clearly needs to be updated anyway...

a-michelis, 2025-01-07:
I'm 90% there, the only issue is the `-m32` part which seems to fail in a compiler level -cmake informs me that _the compiler was not able to compile a simple program_ .

Also, it's the issue is visible through the last triggering of the workflow:

[image]
[image]

I work on this issue on [a-michelis:libstlink/feature/unified_workflow](https://github.com/a-michelis/libstlink/tree/feature/unified_workflow), which has the following workflow result:

[image]
[image]
[image]

Nightwalker-87, 2025-01-07:
> ```c
> value = (int32_t)strtol(str, &tail, 10);
> ```

--> FMPOV the uint16_t range should do for that task in `stlink-lib/helper.c`.

a-michelis, 2025-01-07:
> I intend to give it a try as well by running the build tests locally. Edit: I have fixed the env-issue on the build tasks and added the appropriate parametres. You can focus on the `-m32` issue.

According to my local tests LIBUSB is the issue here - ubuntu provides it only as an x64 binary. Rebuilding and installing it separatelly using `-m32` (through `libusb-cmake` repo) makes stlink to build properly with `-m32`. I'll keep my env changes for now and we can de-conflict our differences later.

a-michelis, 2025-01-07:
Some updates: 

1. I merged all linux based approach to one matrix job for better readability and DRY config.

2. 32-bit versions download and install libusb from `libusb-cmake` (successfully)

3. There seems to be an issue with gtk3 on 32-bit builds which I don't know If i'm able to fix (or how to exlcude gui from the build, if not necessary on 32-bit versions)

Nightwalker-87, 2025-01-07:
> 3\. There seems to be an issue with gtk3 on 32-bit builds which I don't know If i'm able to fix (or how to exclude gui from the build, if not necessary on 32-bit versions)

Simply remove the `libgtk-3-dev` package dependency according to `compiling.md`. The `cmake` routine will then skip building the gui. We may look at this separately later on.
I propose that you start an new PR collecting your improvements to allow for a review. I can then give some comments and compare it with my attempts on this topic. The idea of grouping the workflow jobs into a matrix sounds good. :smiley:

a-michelis, 2025-01-07:
# Behold: the **B̵̗̺͕͈̜̀̂̔Ë̴̬͔̭́̕͠͠H̷͖͙̠̞̯͓͈̻̙͖̞̳̗̋͌̚Ę̵̛͖̳̗̇̉̏͒́̈́͛̿̓͝ͅͅͅM̶̛̛̭̳̪͕̌͒̏͐͑͛̌́͘O̵̦̱̳͎͕̙͒̈́͑̋͛͒̐̑̚͠Ť̷̨̻̤͚̺̠̝͍̯͔̝̦͎͍̬̒́͂̀̑̓̄̐ͅͅH̶̢̧̜͉͉̥͎̟͖̰̼̳̗̦͖̥̦͗̇́̍͛́̍͗͆͐̇̉̚͘̚̚͝͠**

[image]

## Comments and Observations:

- All 32-bit clang compilations fail
- 20.04 32-bit compilations fail due to some kind of missing refference to libpthread 
[image]
- All 64-bit compilations pass in all combos.

a-michelis, 2025-01-08:
We're on the right track.

Only Ubuntu 20.04 32 bit fails now (both gcc/clang), due to not being able to link to pthread

[image]

a-michelis, 2025-01-09:
> I've noticed failing C/C++ CI Tests on Ubuntu during `test-flash`. No idea on that yet, as `make debug` passed successfully.

Noticed the same thing - I rebased my branch to this commit and now all tests are failing. I will revert to the old state to finish the pipeline and then pick up the updates.

As for the #1445 fix, we again forgot to add the explicit typecatsting for clang (strtol returns u64, while our variable is u32, and this makes the unnecessarily pedantic clang to fail on release build, without explicit casting)

Nightwalker-87, 2025-01-09:
You're right, I had it in my original version though. :see_no_evil: As it affects CI building, can you place that small bit in your part, together with the other mentioned changes for `pthread`? You may also squash at least some of the single commits, before creating a PR later on, I propose.

a-michelis, 2025-01-09:
> You're right, I had it in my original version though. 🙈 As it affects CI building, can you place that small bit in your part, together with the other mentioned changes for `pthread`? You may also squash at least some of the single commits, before creating a PR later on, I propose.

I won't be blindly PR-ing my commit mess! 🤣 I will checkout a new branch and re-commit the changes in a more structured fashion, so we can actually watch back and understand what happened in the future. 

I haven't found a solution about pthread yet, as i didn't do anything differently, specifically on ubuntu 20.04. Still searching for clues!

As for the changes, I believe i can commit the whole new function for the frequency, as it seems solid enough and easily  extensible in the future (aka ST-Link V4 goes GHz range 🤣), together with the explicit typecasting! As for the rest, I haven't dived that far (yet) to know if I'm doing right or wrong things in the code.

What's peculliar, tho, is that, with your current commit, all test succeed localy (24.04 wsl + msvc) - at least the ones that were previously passing- and the same happened when using full-blown ubuntu runners on nektos act (sweet jesus, these images are monsters).

Nightwalker-87, 2025-01-09:
> As for the rest, I haven't dived that far (yet) to know if I'm doing right or wrong things in the code.

As nobody really has or ever had (in a whole) - you never walk alone :musical_note: :rofl:
Nevertheless, one may find it impressive how stable the toolset behaves in obviously ordinary use scenarios. :eyes:
However, I must admit that there also were some significant (structural) improvements in the recent years.

I'll see what I can do on the Nucleo-H563ZI-Issue in the meanwhile...

Nightwalker-87, 2025-01-09:
I found the error: 
1) The `windows MSVC` CI does not run `make test`, which causes the fail.
2) What fails exactly is the frequency parsing. Why? - because "no prefix" denotes **kHz** (default) and not Hz. This is a mistake from commit 11e357ae2a34c5f5911c0051fa513659b9bbd7fa. Thus the value ranges don't match the expected ones. This does not only need fixing but also inline documentation.

I'll take care of it and hence also address the typecasting in that file needed for clang compilation to pass.

a-michelis, 2025-01-09:
> As nobody really has or ever had (in a whole) - you never walk alone :musical_note: :rofl:

a-michelis, 2025-01-09:
Now, to recap:

1. I've changed all linux jobs to 2 distinct matrices - 1 for 64-bit and 1 for 32-bit
2. I corrected the compilers used on each scenario, using a job-wide `env` tag on each job of the matrix-jobs.
3. On 32 bit, I added I removed gtk3 and Libusb from installed dependencies. I also used `libusb-cmake` to build and install libusb in 32-bit mode.
4. I'm currently experimenting with glibc6 and gcc variants, as well as with compiler/linker flags in an attempt to make stlink to build properly (aka to find the flippin 32-bit  pthread lib)

Nightwalker-87, 2025-01-09:
Running `sudo make test` locally following equal procedures to the CI tests gives:

```
Running tests...
Test project /.../stlink/build/Debug
    Start 1: test-usb
1/3 Test #1: test-usb .........................   Passed    0.00 sec
    Start 2: test-sg
2/3 Test #2: test-sg ..........................   Passed    0.01 sec
    Start 3: test-flash
3/3 Test #3: test-flash .......................   Passed    0.00 sec

100% tests passed, 0 tests failed out of 3

Total Test time (real) =   0.02 sec
```

a-michelis, 2025-01-10:
> Running `sudo make test` locally following equal procedures to the CI tests gives:
>
> ```
> Running tests...
> Test project /.../stlink/build/Debug
>     Start 1: test-usb
> 1/3 Test #1: test-usb [...]

exactly, that's what I observed as well!

a-michelis, 2025-01-10:
Finally, Progress!

[image]

On my way to re-add the detailed configurations back to the game, now
Time for B̴͇̈́E̷̦͓̿H̵̜̔É̷̻M̴̮̔̄O̷͉͔̅̐T̸̻͑H̸͖̺̅'s rebirth!

a-michelis, 2025-01-10:
# P A R T Y
[image]

PR #1449 (2025-01-11): [feature] Corrected and unified GitHub-Actions C/C++ CI workflow
As there where many problems with the previous C-Cpp Github actions workflow, a few changes were made:

- All checkout steps were bumped from `v2` to `v4`, as v2 is about to be deprecated
- Instead of individual jobs , two matrices were implemented: one for x32 compilations and one for x64
- Compilation flags and compiler definitions are now properly defined using `env` keys within the workflow
- All `make` calls were replaced with their equivalent `cmake` counterparts, to provide a unified methodology of testing across all platf [... cut]
closed 2025-01-11 by Nightwalker-87