stlink-org/stlink #1454: Can't install stlink_1.8.0-1_amd64.deb on Ubuntu 20.04

snowuyl, 2025-02-27:
**Thank you for giving feedback to the stlink project.**

---

**NOTE: In order to offer sufficient and the best possible support, please read /CONTRIBUTING.md and follow the given instructions _before_ submitting a ticket.**

**Bug reports and/or feature requests will be deleted, if they violate our contribution guidelines and if no issue-template is used!** Thank you for your support.

---

- [ ] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers to isolate and target your respective issue, please take some time to select the check boxes below and fill out each of the following items appropriate to your specific problem.

- [ ] Programmer/board type: [STLINK V2] (e.g STLINK /V1, /V2, /V2-onboard, /V2-clone, /V3)
- [ ] Operating system an version: [Ubuntu 20.04] (e.g Linux, Windows)
- [ ] **stlink tools version** and/or git commit hash: [stlink_1.8.0-1_amd64.deb] (e.g v1.6.1/git-d0416149)
- [ ] stlink commandline tool name: [st-info] (e.g `st-info`, `st-flash`, `st-trace`, `st-util`)
- [ ] Target chip (and board, if applicable): [STM32F103C8T6] (e.g STM32F103C8T6 (NUCLEO-F103RB))

Further we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

Commandline output:
sudo dpkg  -i stlink_1.8.0-1_amd64.deb 
Selecting previously unselected package stlink.
(Reading database ... 167383 files and directories currently installed.)
Preparing to unpack stlink_1.8.0-1_amd64.deb ...
Unpacking stlink (1.8.0-1) ...
dpkg: error processing archive stlink_1.8.0-1_amd64.deb (--install):
 trying to overwrite '/etc/modprobe.d/stlink_v1.conf', which is also in package stlink-tools 1.6.0+ds-1
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Errors were encountered while processing:
 stlink_1.8.0-1_amd64.deb

`short description of the expected value`

Nightwalker-87, 2025-02-27:
Package conflict: `trying to overwrite '/etc/modprobe.d/stlink_v1.conf', which is also in package stlink-tools 1.6.0+ds-1`
Uninstall distributed version and it's dependencies first.

Nightwalker-87, 2025-03-05:
As appropriate advice has been given, I am closing this ticket now as resolved.

closed 2025-03-05 by Nightwalker-87