stlink-org/stlink #1456: STM32G474: programming flash error, non-erased value.

Gnarflord, 2025-03-03:
- [x] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers to isolate and target your respective issue, please take some time to select the check boxes below and fill out each of the following items appropriate to your specific problem.

- [x] Programmer/board type: STLINK V2
- [x] Operating system an version: Linux, Fedora
- [x] **stlink tools version** and/or git commit hash: 1.8.0
- [x] stlink commandline tool name: `st-flash`
- [x] Target chip (and board, if applicable): STM32G474VET

I'm trying to flash a rather big image onto my STM. Unfortunately this is failing:
```
❯ st-flash write my_image.bin 0x8000000
st-flash 1.8.0-94-ge788c54
2025-03-03T10:51:39 INFO common.c: STM32G47x_G48x: 128 KiB SRAM, 512 KiB flash in at least 2 KiB pages.
file my_image.bin md5 checksum: 3a95f08b82d1fecb9f3ecdd2d2d3b676, stlink checksum: 0x02caab08
2025-03-03T10:51:39 INFO common_flash.c: Attempting to write 293428 (0x47a34) bytes to stm32 address: 134217728 (0x8000000)
-> Flash page at 0x8000000 erased (size: 0x800)
-> Flash page at 0x8000800 erased (size: 0x800)
-> Flash page at 0x8001000 erased (size: 0x800)
-> Flash page at 0x8001800 erased (size: 0x800)
[...]
-> Flash page at 0x8045800 erased (size: 0x800)
-> Flash page at 0x8046000 erased (size: 0x800)
-> Flash page at 0x8046800 erased (size: 0x800)
-> Flash page at 0x8047000 erased (size: 0x800)
-> Flash page at 0x8047800 erased (size: 0x800)

2025-03-03T10:51:43 INFO flash_loader.c: Starting Flash write for WB/G0/G4/L5/U5/H5/C0
  1/143 pages written
  2/143 pages written
  3/143 pages written
  4/143 pages written
  5/143 pages written
  6/143 pages written
  7/143 pages written
[...]
141/143 pages written
142/143 pages written
143/143 pages written

2025-03-03T10:52:21 ERROR common_flash.c: Flash memory contains a non-erased value
2025-03-03T10:52:21 ERROR common_flash.c: Flash programming error: 0x000000a0
stlink_fwrite_flash() == -1
❯ 
```
The error looks a lot like #1310, which is related to wrong handling of dual-bank flash devices. Using debug output shows the following:
```
2025-03-03T10:54:39 DEBUG read_write.c: *** stlink_write_debug32 0x200002f9 to 0x08047a30
2025-03-03T10:54:39 DEBUG read_write.c: *** stlink_read_debug32  0x000000a8 at 0x40022010

2025-03-03T10:54:39 DEBUG read_write.c: *** stlink_write_debug32 0000000000 to 0x08047a34
2025-03-03T10:54:39 DEBUG read_write.c: *** stlink_read_debug32  0x000000a8 at 0x40022010
2025-03-03T10:54:39 DEBUG read_write.c: *** stlink_read_debug32  0x000000a8 at 0x40022010
2025-03-03T10:54:39 DEBUG read_write.c: *** stlink_read_debug32  0x000000a8 at 0x40022010
2025-03-03T10:54:39 ERROR common_flash.c: Flash memory contains a non-erased value
2025-03-03T10:54:39 ERROR common_flash.c: Flash programming error: 0x000000a0
2025-03-03T10:54:39 DEBUG read_write.c: *** stlink_read_debug32  0x08001e09 at 0x08000004
2025-03-03T10:54:39 DEBUG read_write.c: *** stlink_write_reg
2025-03-03T10:54:39 DEBUG common_legacy.c: *** stlink_run ***
2025-03-03T10:54:39 DEBUG read_write.c: *** stlink_read_reg
2025-03-03T10:54:39 DEBUG read_write.c:  (16) ***
2025-03-03T10:54:39 DEBUG common_legacy.c: data_len = 8 0x8
 80 00 89 4b 00 00 00 61
2025-03-03T10:54:39 DEBUG usb.c: r_idx (16) = 0x61000000
stlink_fwrite_flash() == -1
2025-03-03T10:54:39 DEBUG common_legacy.c: *** stlink_exit_debug_mode ***
2025-03-03T10:54:39 DEBUG read_write.c: *** stlink_write_debug32 0xa05f0000 to 0xe000edf0
2025-03-03T10:54:39 DEBUG common_legacy.c: *** stlink_close ***
```
The error only appears if the device was written to previously. If I first erase the device with `st-flash erase` the next `st-flash write` will succeed:
```
❯ st-flash erase
st-flash 1.8.0-94-ge788c54
2025-03-03T10:55:31 INFO common.c: STM32G47x_G48x: 128 KiB SRAM, 512 KiB flash in at least 2 KiB pages.
Mass erasing...
Mass erase completed successfully.

❯ st-flash write my_image.bin 0x8000000
st-flash 1.8.0-94-ge788c54
2025-03-03T10:55:36 INFO common.c: STM32G47x_G48x: 128 KiB SRAM, 512 KiB flash in at least 2 KiB pages.
file my_image.bin md5 checksum: 3a95f08b82d1fecb9f3ecdd2d2d3b676, stlink checksum: 0x02caab08
2025-03-03T10:55:36 INFO common_flash.c: Attempting to write 293428 (0x47a34) bytes to stm32 address: 134217728 (0x8000000)
-> Flash page at 0x8000000 erased (size: 0x800)
-> Flash page at 0x8000800 erased (size: 0x800)
[...]
142/143 pages written
143/143 pages written

2025-03-03T10:56:19 INFO common_flash.c: Starting verification of write complete
2025-03-03T10:56:21 INFO common_flash.c: Flash written and verified! jolly good!
2025-03-03T10:56:21 INFO common.c: Go to Thumb mode
❯ 
```

I also tried using st-util together with gdb-server. Interestingly this fails exactly at address 0x804000:
```

st-util -p 50000 --no-reset
st-util 1.8.0
2025-03-03T10:10:35 INFO common.c: STM32G47x_G48x: 128 KiB SRAM, 512 KiB flash in at least 2 KiB pages.
2025-03-03T10:10:35 INFO gdb-server.c: Listening at *:50000...
2025-03-03T10:10:35 INFO common.c: STM32G47x_G48x: 128 KiB SRAM, 512 KiB flash in at least 2 KiB pages.
2025-03-03T10:10:35 INFO gdb-server.c: Found 6 hw breakpoint registers
2025-03-03T10:10:35 INFO gdb-server.c: GDB connected.
2025-03-03T10:10:35 INFO gdb-server.c: Found 6 hw breakpoint registers
2025-03-03T10:10:36 INFO common.c: STM32G47x_G48x: 128 KiB SRAM, 512 KiB flash in at least 2 KiB pages.
2025-03-03T10:10:36 INFO gdb-server.c: flash_erase: block 08030000 -> 18000
2025-03-03T10:10:36 INFO gdb-server.c: flash_erase: page 08030000
2025-03-03T10:10:36 INFO gdb-server.c: flash_erase: page 08030800
[...]
2025-03-03T10:10:37 INFO gdb-server.c: flash_erase: page 08046000
2025-03-03T10:10:37 INFO gdb-server.c: flash_erase: page 08046800
2025-03-03T10:10:37 INFO gdb-server.c: flash_erase: page 08047000
2025-03-03T10:10:37 INFO gdb-server.c: flash_erase: page 08047800
2025-03-03T10:10:37 INFO flash_loader.c: Starting Flash write for WB/G0/G4/L5/U5/H5/C0
2025-03-03T10:10:37 INFO gdb-server.c: flash_do: block 08030000 -> 18000
2025-03-03T10:10:37 INFO gdb-server.c: flash_do: page 08030000
  1/1   pages written

2025-03-03T10:10:38 INFO gdb-server.c: flash_do: page 08030800
  1/1   pages written

2025-03-03T10:10:38 INFO gdb-server.c: flash_do: page 08031000
  1/1   pages written
[...]
2025-03-03T10:10:45 INFO gdb-server.c: flash_do: page 0803f000
  1/1   pages written

2025-03-03T10:10:46 INFO gdb-server.c: flash_do: page 0803f800
  1/1   pages written

2025-03-03T10:10:46 INFO gdb-server.c: flash_do: page 08040000
  1/1   pages written

2025-03-03T10:10:46 ERROR common_flash.c: Flash memory contains a non-erased value
2025-03-03T10:10:46 ERROR common_flash.c: Flash programming error: 0x000000a0
Receive signal 15. Exiting...
2025-03-03T10:10:46 ERROR usb.c: READREG send request failed: LIBUSB_ERROR_BUSY
2025-03-03T10:10:46 INFO common.c: Go to Thumb mode
2025-03-03T10:10:46 ERROR usb.c: WRITEREG send request failed: LIBUSB_ERROR_BUSY
2025-03-03T10:10:46 ERROR usb.c: WRITEDEBUGREG send request failed: LIBUSB_ERROR_BUSY
2025-03-03T10:10:46 ERROR usb.c: DEBUG_EXIT send request failed: LIBUSB_ERROR_BUSY
```
I suspect that erasing the flash pages of the second bank silently fails but I don't have enough knowledge about STMs or stlink to further debug this issues.

Ant-ON, 2025-03-03:
@Gnarflord Yes, most likely the pages of the second bank are not erased. The application does not support the two-bank option for all series...

Ant-ON, 2025-03-03:
@Gnarflord Yes, you right. The `BKER` flags is different for G0 and G4

PR #1457 (2025-03-05): [feature] STM32G4: Erase pages on flash bank 2
Fixes #1456 

I added a further define to differentiate between the bit positions of BKER on G0 and G4 devices. Apparently bank 2 is not even supported for the G0 at all, so this BKER field was previously never correctly used? Someone with more experience should definitely confirm that my changes do not break support for other devices..
closed 2025-03-05 by Nightwalker-87