stlink-org/stlink #1461: Failed to build stlink on Ubuntu 20.04

snowuyl, 2025-03-30:
**Thank you for giving feedback to the stlink project.**

---

**NOTE: In order to offer sufficient and the best possible support, please read /CONTRIBUTING.md and follow the given instructions _before_ submitting a ticket.**

**Bug reports and/or feature requests will be deleted, if they violate our contribution guidelines and if no issue-template is used!** Thank you for your support.

---

- [ ] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers to isolate and target your respective issue, please take some time to select the check boxes below and fill out each of the following items appropriate to your specific problem.

- [ ] Programmer/board type: [STLINK /V2] (e.g STLINK /V1, /V2, /V2-onboard, /V2-clone, /V3)
- [ ] Operating system an version: [Ubuntu 20.04] (e.g Linux, Windows)
- [ ] **stlink tools version** and/or git commit hash: [git master] (e.g v1.6.1/git-d0416149)
- [ ] stlink commandline tool name: [st-info] (e.g `st-info`, `st-flash`, `st-trace`, `st-util`)
- [ ] Target chip (and board, if applicable): [STM32F103C8T6] (e.g STM32F103C8T6 (NUCLEO-F103RB))

Further we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

Commandline output:
sudo apt update
sudo apt-get install git make cmake libusb-1.0-0-dev
sudo apt-get install gcc build-essential
git clone https://github.com/stlink-org/stlink
cd stlink
cmake .
make
OUTPUT/ERROR of the commandline tool(s)
`Hit:1 http://tw.archive.ubuntu.com/ubuntu focal InRelease                   
Hit:2 http://tw.archive.ubuntu.com/ubuntu focal-updates InRelease           
Hit:3 http://tw.archive.ubuntu.com/ubuntu focal-backports InRelease
Hit:4 http://security.ubuntu.com/ubuntu focal-security InRelease
Reading package lists... Done                          
Building dependency tree       
Reading state information... Done
375 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  dh-strip-nondeterminism dwz intltool-debian libarchive-cpio-perl libarchive-zip-perl libc-dev-bin
  libcrypt-dev libdebhelper-perl libfakeroot libfile-stripnondeterminism-perl libltdl-dev
  libmail-sendmail-perl libsub-override-perl libsys-hostname-long-perl linux-libc-dev po-debconf
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  git-man liberror-perl
Suggested packages:
  git-daemon-run | git-daemon-sysvinit git-doc git-el git-email git-gui gitk gitweb git-cvs
  git-mediawiki git-svn
The following NEW packages will be installed:
  git git-man liberror-perl
0 upgraded, 3 newly installed, 0 to remove and 375 not upgraded.
Need to get 5526 kB of archives.
After this operation, 38.8 MB of additional disk space will be used.
Do you want to continue? [Y/n] 
Get:1 http://tw.archive.ubuntu.com/ubuntu focal/main amd64 liberror-perl all 0.17029-1 [26.5 kB]
Get:2 http://tw.archive.ubuntu.com/ubuntu focal-updates/main amd64 git-man all 1:2.25.1-1ubuntu3.14 [887 kB]
Get:3 http://tw.archive.ubuntu.com/ubuntu focal-updates/main amd64 git amd64 1:2.25.1-1ubuntu3.14 [4613 kB]
Fetched 5526 kB in 5s (1206 kB/s)
Selecting previously unselected package liberror-perl.
(Reading database ... 151886 files and directories currently installed.)
Preparing to unpack .../liberror-perl_0.17029-1_all.deb ...
Unpacking liberror-perl (0.17029-1) ...
Selecting previously unselected package git-man.
Preparing to unpack .../git-man_1%3a2.25.1-1ubuntu3.14_all.deb ...
Unpacking git-man (1:2.25.1-1ubuntu3.14) ...
Selecting previously unselected package git.
Preparing to unpack .../git_1%3a2.25.1-1ubuntu3.14_amd64.deb ...
Unpacking git (1:2.25.1-1ubuntu3.14) ...
Setting up liberror-perl (0.17029-1) ...
Setting up git-man (1:2.25.1-1ubuntu3.14) ...
Setting up git (1:2.25.1-1ubuntu3.14) ...
Processing triggers for man-db (2.9.1-1) ...
Cloning into 'stlink'...
remote: Enumerating objects: 14390, done.
remote: Counting objects: 100% (2182/2182), done.
remote: Compressing objects: 100% (397/397), done.
remote: Total 14390 (delta 1986), reused 1788 (delta 1785), pack-reused 12208 (from 3)
Receiving objects: 100% (14390/14390), 17.26 MiB | 985.00 KiB/s, done.
Resolving deltas: 100% (9491/9491), done.
-- The C compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
CMake Error in /home/andyueng/samba/Backup/Software/Ubunutu/20.04/stlink/CMakeFiles/CMakeTmp/CMakeLists.txt:
  Target "cmTC_c9580" requires the language dialect "C17" (with compiler
  extensions), but CMake does not know the compile flags to use to enable it.

CMake Error at /usr/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:44 (try_compile):
  Failed to generate test project build system.
Call Stack (most recent call first):
  CMakeLists.txt:71 (project)

-- Configuring incomplete, errors occurred!
See also "/home/andyueng/samba/Backup/Software/Ubunutu/20.04/stlink/CMakeFiles/CMakeOutput.log".
See also "/home/andyueng/samba/Backup/Software/Ubunutu/20.04/stlink/CMakeFiles/CMakeError.log".
-- The C compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
CMake Error in /home/andyueng/samba/Backup/Software/Ubunutu/20.04/stlink/CMakeFiles/CMakeTmp/CMakeLists.txt:
  Target "cmTC_3a91c" requires the language dialect "C17" (with compiler
  extensions), but CMake does not know the compile flags to use to enable it.

CMake Error at /usr/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:44 (try_compile):
  Failed to generate test project build system.
Call Stack (most recent call first):
  CMakeLists.txt:71 (project)

-- Configuring incomplete, errors occurred!
See also "/home/andyueng/samba/Backup/Software/Ubunutu/20.04/stlink/CMakeFiles/CMakeOutput.log".
See also "/home/andyueng/samba/Backup/Software/Ubunutu/20.04/stlink/CMakeFiles/CMakeError.log".
make: *** [Makefile:58: build/Release] Error 1

DebugBSD, 2025-03-30:
It seems that you C compiler does not support the standard C17. I'm using Fedora 40, Ubuntu 24.04 and OpenBSD (-CURRENT) and I didn't have any problem building from source code. 

Are you sure your compiler support C17? Can you check it out?

snowuyl, 2025-03-30:
Thanks for your reply! I will check it.

snowuyl, 2025-03-30:
unzip test_c++17.zip
$ g++ -std=c++17 test_c++17.cpp -o test_c++17
$ ./test_c++17 
Optional value: 42
$ g++ --version
g++ (Ubuntu 9.4.0-1ubuntu1~20.04.2) 9.4.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE

snowuyl, 2025-03-30:
[

[test_c++17.zip](https://github.com/user-attachments/files/19525330/test_c%2B%2B17.zip)

](url)

snowuyl, 2025-03-30:
cat test_c++17.cpp 
#include <optional>
#include <iostream>

int main() {
    std::optional<int> opt = 42;
    if (opt) {
        std::cout << "Optional value: " << *opt << std::endl;
    }
    return 0;
}

snowuyl, 2025-03-30:
$ gcc -std=c17 test_c17.c -o test_c17
$ ./test_c17 
C17 is supported!
$ gcc --version
gcc (Ubuntu 9.4.0-1ubuntu1~20.04.2) 9.4.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

snowuyl, 2025-03-30:
cat test_c17.c 
#include <stdio.h>

_Static_assert(sizeof(int) == 4, "Integers must be 4 bytes");

int main() {
    printf("C17 is supported!\n");
    return 0;
}

DebugBSD, 2025-03-30:
Is it possible to use a more recent version of CMake? Maybe a version >=3.21? 

For sure the problem is not about the source code, I thought that maybe your compiler does not support C17 but you have shown that gcc 9 supports it so, please, try to install a more recen version of cmake. I'm using 3.28 in Ubuntu.

snowuyl, 2025-03-30:
Thanks for your reply! I will test it later!

closed 2025-04-02 by snowuyl