stlink-org/stlink #1472: 32F746GDISCOVERY: st-util.exe doesn't run

singds, 2025-06-06:
---

- [x] I made serious effort to avoid creating duplicate or nearly similar issue
- [x] Programmer/board type: ST-LINK/V2-1
- [x] Operating system an version: Windows 11 Pro
- [x] **stlink tools version** and/or git commit hash: v1.8.0
- [x] stlink commandline tool name: `st-util`
- [x] Target chip (and board, if applicable): 32F746GDISCOVERY

I download Release v1.8.0 and extract it.
Then in powershell:
```
> cd .\stlink-1.8.0-win32\bin\
> st-util.exe
>
```

There is no output. When I try to launch any of the tools, they close printing nothing. The previous version v1.7.0 works as expected.

a-michelis, 2025-07-24:
Let's see:

- If you run 
  ```powershell
  cd .\stlink-1.8.0-win32\bin\
  st-util.exe
  echo $?
  ```
  Do you get `True` or `False`?

- do you have `libusb-1.0` installed and available in the `Path` environment variable of your computer?

jaycee1980, 2025-09-05:
Just tried this myself and sure enough the 1.8.0 executables do not run. 1.7.0 does.
If you double click one of the executables from the desktop, you get a dialog that says

"The application was unable to start correctly (0xc000007b). Click OK to close the application."

This is caused by a missing libusb-1.0.dll. To solve this download libusb from https://github.com/libusb/libusb and copy the DLL from MinGW64\dll\libusb-1.0.dll into the same path as the executables

closed 2026-09-17 by Nightwalker-87