stlink-org/stlink #1474: [feature] New logging system in preparation for an intended stlink library approach

Nightwalker-87, 2025-07-16:
This issue attempts to give STLink project a proper logging system in preparation for a library approach, to enable users incorporate stlink-lib into their projects.
Proposal: Main changes (in a nutshell)

- The project may use [libusb-cmake](https://github.com/libusb/libusb-cmake) project as its libusb provider, which in turn allows for custom-built libusb, tailored to each system's needs.
- All included headers may be placed into appropriate stlink sub-folders that are included into the project as system includes, to allow for system-like includes. This change requires the change of all inclusions in the project to change from #include "%HEADER%.h" to #include <stlink/%HEADER%.h>
- Several headers have exposed functionality which could be used by custom software, using extern "C" blocks.

Much needed prior additions that are missing:

1. Changing the logging infrastructure to allow programmatically managed logging, instead of STDOUT logs.
2. Help to accommodate the appropriate cmake modifications for other BSD/*NIX based systems, as well as verify/implement a proper handling of pre-installed libusb followed by exhaustive testing
3. Documentation of the exported functions (and of course evaluate whether the overall exposure is excessive or too limited).

A.M.

a-michelis, 2025-07-18:
We're getting somewhere:

[image]

a-michelis, 2025-07-24:
As seen in the pictures above, I managed to implement the Logging system itself (as a separate shared library) , but I have some questions about the inner workings of ST-Link device instances:

1. Each ST-Link instance seem to contain its own verbosity level. That said, `open` implementations, seem to call a _global_ (?) initializer. Does the current logging facility only support global logging and this field was added as a placeholder, Or there are distinctions from instance to instance that I missed?

2. If no per-instance logging level is supported, would that be something we're interested in?

> #### Sidenote:
> 
> As for the C++ re-implementation, the more I dive into the code, the more C++ makes sense to me - It'd make the codebase so much more understandable and maintainable... I'm probably going to give it a try early Aug!

Nightwalker-87, 2025-08-08:
> 1. Each ST-Link instance seem to contain its own verbosity level. That said, `open` implementations, seem to call a _global_ (?) initializer. Does the current logging facility only support global [...]

I assume only global logging is the case, but I'm not definitely sure.

>     2. If no per-instance logging level is supported, would that be something we're interested in?

Possibly not, unless there is a specific use case  you can imagine that may be worth a discussion.

> > As for the C++ re-implementation, the more I dive into the code, the more C++ makes sense to me - It'd make the codebase so much more understandable and maintainable... I'm probably going to give it a try early Aug!

Much appreciated. Is there any specific basic functionality that would make sense to start with in a step-by-step approach?

I can grant some extended permissions in the project to ease contribution for you.

a-michelis, 2025-09-18:
I hope you solve the issue soon!

I just pushed a first rough boilerplate cpp abstract class for ST-Link (on my fork, @ [cpp_rewrite](https://github.com/a-michelis/libstlink/tree/cpp_rewrite) branch). I need to revisit the exposed (virtual) functions (having guidance as to what shall be exposed would be helpfull there), and I might tone it down a bit with the modernization for easier transition, but I can see this working!

Nightwalker-87, 2026-08-16:
Updated key focus for this feature request:

- Several headers have exposed functionality which could be used by custom software, using extern "C" blocks.
- **Changing the logging infrastructure to allow programmatically managed logging, instead of STDOUT logs.**
- Documentation of the exported functions (and of course evaluate whether the overall exposure is excessive or too limited).

-----

### General (coding) conventions:
  - Align to License Terms of the project
  - intended coding guidelines: https://www.perforce.com/blog/kw/NASA-rules-for-developing-safety-critical-code

### Formatting:
  - use 4 spaces instead of tabs
  - indentation: 4 spaces
    - also minimal distance from code-line ending to the beginning of inline comments
    - also minimal distance between @ tags with parametres and description in header comments for functions
  - open decision: text width in characters per line
  - header comments: nice formatting in general; however I suppose to align the "@return" parametres as well
  - unify { } formatting in functions: I prefer "1TBS / K&R" over "Classic K&R".
  - 2 empty lines between block of includes and beginning of code
  - 2 empty lines between file header and #ifndef (in case of header files) or block of includes (in case of source files)
  - continued use of unified file header
  - use UTF-8 character encoding and \LF break type (Unix)

### Refactoring:
  - I really like the idea to introduce a source/header file pair covering all hardware-related functionalities of compatible STLink devices. As to my knowledge this content is currently spread across several code sections and maybe even across different source files. Also there is no clear distinction between several hardware variants, which should be directly discoverable in the source code. These are:

    - STLINK/V1
    - STLINK/V2, /V2-A, /V2-B, /V2EC
    - STLINK/V2-1
    - STLINK-V3SET, -V3MODS, -V3MINI, -V3MINIE, -V3E, -V3EC, -V3PWR

  - Continued refactoring of source files `common_legacy.c` and `sg_legacy.c`

Please feel free to add further ideas in this context.

Nightwalker-87, 2026-10-04:
## Findings on `testing` (`876a50f`) regarding the library goals of this issue

I checked the current `testing` head against the goals of this issue. In short, the logging part is a small, self-contained change. The library/API side has more open points than are listed here, and one of them is an ABI break that already shipped in v1.9.0.

### Logging

1. **Output is hard-wired to stderr.** `ugly_log()` writes directly to stderr and calls `fflush(stdout)` first ([logging.c#L26-L31](https://github.com/stlink-org/stlink/blob/876a50f547bc13e04a1144866f5fea3a4d6b4fe5/src/stlink-lib/logging.c#L26-L31)), so it touches the host application's stdout. The log level is a single global variable that defaults to `UDEBUG`.
2. **The library overrides the application's log level.** `stlink_open_usb()`, `stlink_open_remote()` and `stlink_v1_open_inner()` each call `ugly_init(verbose)` ([usb.c#L1446](https://github.com/stlink-org/stlink/blob/876a50f547bc13e04a1144866f5fea3a4d6b4fe5/src/stlink-lib/usb.c#L1446), [remote.c#L394](https://github.com/stlink-org/stlink/blob/876a50f547bc13e04a1144866f5fea3a4d6b4fe5/src/stlink-lib/remote.c#L394), [sg_legacy.c#L1074](https://github.com/stlink-org/stlink/blob/876a50f547bc13e04a1144866f5fea3a4d6b4fe5/src/stlink-lib/sg_legacy.c#L1074)), so opening a device resets a process-wide setting. st-info and stlink-gui pass `0`, which suppresses every message including errors. st-server has to call `ugly_init()` a second time to restore its own level ([server.c#L151-L160](https://github.com/stlink-org/stlink/blob/876a50f547bc13e04a1144866f5fea3a4d6b4fe5/src/st-server/server.c#L151-L160)).
3. **79 direct `printf`/`fprintf`/`perror` calls in stlink-lib bypass the logger**, and many of them write to stdout. Most are in common_legacy.c (18), common_flash.c (14), chipid.c (14) and flash_loader.c (13); the rest are in remote.c, md5.c, map_file.c, read_write.c, option_bytes.c and usb.c. They fall into three groups:
   - errors and diagnostics ("Failed to parse chip-id", "addr too low") → `ELOG`/`WLOG`/`DLOG`
   - progress ("x/y pages written", "Mass erasing...") → a separate progress callback, since a GUI wants a progress bar, not log lines
   - results (MD5 checksum in md5.c, option byte value in option_bytes.c) → returned to the caller, and the tool does the printing
4. **Lines can interleave.** Each log line is written by three separate `fprintf` calls (timestamp, level, message). Since `st-info --probe` now probes in parallel (#1483), lines from different threads can get mixed. Formatting into one buffer and writing it once fixes this.
5. **The spdlog scaffolding is dead code but still gets installed.** This covers `logging_spdlog_wr.h` and the `SPDLOG_LOGGING` blocks; the define itself is commented out.
6. **libusb output bypasses everything.** libusb >= 1.0.24 is now the minimum, so `libusb_set_log_cb()` is available to route libusb messages through the same sink. FreeBSD's own libusb implementation needs checking.

I agree with @jameswalmsley that a plain C callback is enough, for example:

```c
typedef void (*stlink_log_fn)(void *user, int level, const char *tag, const char *msg);
void stlink_set_log_handler(stlink_log_fn fn, void *user); /* NULL restores the default stderr output */
void stlink_set_log_level(int level);
```

The 368 `DLOG`/`ILOG`/`WLOG`/`ELOG` call sites in stlink-lib (plus 180 in the tools) stay as they are; only `logging.c` changes. Per-instance context, i.e. @a-michelis' use case with several devices at once, doesn't need a C++ rewrite either. `stlink_t` already has a `verbose` field, and the callback can optionally receive the `stlink_t *` (or the serial). This can be introduced gradually, e.g. with an `SLOG(sl, ...)` macro in usb.c and remote.c.

### Library / public API

- **v1.9.0 breaks the ABI without changing the soname.** `uint8_t ap` was inserted in the middle of `struct _stlink` ([stlink.h#L223](https://github.com/stlink-org/stlink/blob/876a50f547bc13e04a1144866f5fea3a4d6b4fe5/inc/stlink.h#L223)), which shifts all following fields. The library is still `libstlink.so.1` (`SOVERSION ${PROJECT_VERSION_MAJOR}`, [CMakeLists.txt#L143](https://github.com/stlink-org/stlink/blob/876a50f547bc13e04a1144866f5fea3a4d6b4fe5/CMakeLists.txt#L143)), so binaries built against the 1.8 headers that access `stlink_t` members will read wrong offsets. Debian ships this library as `libstlink1` and tracks it with a symbols file, so distro packages are affected too. `testing` adds two more fields (`flash_secure`, `secure_csw`, [stlink.h#L260-L261](https://github.com/stlink-org/stlink/blob/876a50f547bc13e04a1144866f5fea3a4d6b4fe5/inc/stlink.h#L260-L261)). They are appended at the end, which keeps existing offsets intact, but the public struct keeps growing with every feature. Short term, `SOVERSION` should be bumped whenever a public struct changes incompatibly. Long term, `stlink_t` should become opaque with accessor functions.
- **Every header gets installed.** `file(GLOB ...)` picks up all headers in `inc/`, `src/stlink-lib/` and `src/stlink-env/` ([CMakeLists.txt#L106](https://github.com/stlink-org/stlink/blob/876a50f547bc13e04a1144866f5fea3a4d6b4fe5/CMakeLists.txt#L106)). Internals like `logging.h`, `sg_legacy.h` or `remote.h` (which pulls in `windows.h` / POSIX socket headers) therefore become public API. We need an explicit list of public headers.
[... cut ...] ake package config, and the removal of the unused SCSI backend in `sg_legacy.c` and of the PID macros from the public headers.

### Already done / outdated in this issue

- libusb-cmake is no longer used. MSVC builds now use vcpkg (#1507), the MinGW cross build fetches upstream libusb via FetchContent, and `Findlibusb.cmake` has dedicated FreeBSD/OpenBSD branches. The task list above could be updated accordingly.

### Further ToDos:

- Still open: refactoring `common_legacy.c` and `sg_legacy.c`.

Nightwalker-87, 2026-10-04:
Commit d464cd69cc721c062e34ede035cf9e9acccff0fc addresses 1) and 2) FMPOV.
Ideas and thoughts on 3) as well as on refactoring of  `common_legacy.c` and `sg_legacy.c` or other related issues are warmly welcomed. 4) will not be part of an upcoming Point-Release.

Nightwalker-87, 2026-10-06:
>  Documentation of the exported functions (and of course evaluate whether the overall exposure is excessive or too limited).

This should be covered by commits b7a27bda980d7ace513d6fcef70c6060bb82f9cb and 6d4d4b3cad7375ee52f01d427993128e618c1289.

Nightwalker-87, 2026-10-06:
# API exposure evaluation {#api_exposure}

Evaluation of the functions exported by stlink-lib, as asked for in point 3 of the issue desciption: is the exposure excessive or too limited? Basis: branch `testing` at 010c74b (v1.9.1-dev), shared library built on Linux.

Method: the exported functions are the defined text symbols of `libstlink.so` (`nm -D --defined-only`). For each one the references in `src/`, `tests/` and the GUI were counted, ignoring comments, string literals and the definition itself; uses through macros (DLOG() etc.) were added by hand.

## Summary

| | Functions |
|---|---|
| Exported by `libstlink.so` | 216 |
| Used by the tools or the GUI, or declared in `inc/` | 71 |
| Used only by the tests | 16 |
| Used across modules of the library only | 33 |
| Used only in the source file defining them | 91 |
| Not used at all | 5 |

| Recommendation | Functions |
|---|---|
| Public: keep, documented in the public API | 68 |
| Public, revise: keep the functionality, change before declaring the API stable | 13 |
| Deprecated: keep for now, already replaced | 3 |
| Internal: used by several modules, hide from the API | 40 |
| Static: used in one source file only | 58 |
| Remove: dead code or the legacy ST-LINK/V1 backend | 33 |
| Move to the tools | 1 |

**Result: the exposure is clearly excessive** – 132 of 216 exported functions (61 %) are not meant for applications – **and at the same time too limited in a few places** that applications need (trace, buffer based memory access, access to the target description without the struct, error details).

## Too much is exposed

1. **No symbol visibility control.** Every non-static function is exported: there is no `-fvisibility=hidden` and no export macro, and MSVC builds use `WINDOWS_EXPORT_ALL_SYMBOLS`. This includes the backend implementations (`_stlink_usb_*`, `_stlink_sg_*`), which are only reachable through stlink_backend_t.
2. **Missing `static`.** 91 functions are used in their own source file only, among them all backend functions, the option byte functions per family and `_parse_version()`, `stlink_jtag_reset()`, `stlink_soft_reset()`.
3. **Generic names in the global namespace.** `send_recv`, `map_file`, `check_file`, `lock_flash`, `time_ms`, `read_uint32`, `Md5Initialise`, `init_chipids`, `process_chipfile`, `search_for_chips`, `calculate_*` and macros like `DLOG`, `ELOG`, `BANK_1`, `FLASH_OBR` can collide with symbols of an application, in particular when it links the static library.
4. **All headers are installed**, including internal ones (`usb.h` with `struct stlink_libusb`, `sg_legacy.h`, `progress.h`, `lib_md5.h`, `stlink_cmd.h`, the register maps `stm32_register.h` and `stm32_flash.h`), flat into `include/stlink` – and directly into `include` on Windows. A header called `usb.h` clashes with the one of libusb-compat. The `<stlink/...>` layout proposed previously fixes this.
5. **The handle is not opaque.** `struct _stlink` is fully public, including the 100 KiB data buffer, the backend pointer and internal state. Applications read the target description from it, and the memory functions pass data through `q_buf`. Every new field changes the ABI: v1.9.0 (field `ap`) and, on `testing`, the TrustZone fields and the progress handler changed the layout while the soname stayed `libstlink.so.1`.
6. **Internal functions in the main header.** `stlink.h` declares `stlink_chip_id()` and `stlink_core_stat()` (both documented as "do not call directly"), `stlink_print_data()` and `stlink_write_buffer_to_sram()`. `stlink_enter_jtag_mode()` was declared without being defined; the declaration is removed with this change.
7. **Tool code in the library.** `arg_parse_freq()` parses command line arguments.
8. **Dead and broken code.** `stlink_set_hw_bp()` (writes its arguments into the data buffer instead of the command, so it never worked), `stlink_clr_hw_bp()`, `stlink_write_dreg()` and `Md5Calculate()` are unused. The complete legacy ST-LINK/V1 backend (`sg_legacy.c`, 31 functions) is only used by the manual hardware test `tests/sg_legacy.c`: `stlink_open_usb()` handles the ST-LINK/V1 itself (protocol 1 in `usb.c`).

## Too little is exposed

1. **SWO trace.** st-trace defines `stlink_trace_enable()`, `stlink_trace_disable()` and `stlink_trace_read()` itself and calls the backend directly (`sl->backend->trace_*`). These belong into the library.
2. **Buffer based memory access.** Reading or writing memory requires copying through `sl->q_buf`, splitting into blocks (6 KiB reads, 64/512 bytes for 8 bit writes) and handling alignment. A `stlink_read_mem(sl, addr, buf, len)` / `stlink_write_mem(...)` pair would hide all of this and allow an opaque handle.
3. **Target description without the struct.** Accessors or a copy-out struct (flash base/size/page size, SRAM, option bytes, OTP, chip id, device name, serial, ST-LINK version) are needed before the handle can become opaque.
4. **Error details.** Failures return -1; the reason only goes to the log. An error code (enum) or `stlink_last_error(sl)` would let applications react.
5. **Device enumeration without connecting.** stlink_probe_usb() opens every ST-LINK and connects to (and possibly resets) its target. A list of the connected ST-LINKs (serial, version, bus/address) and opening one by bus/address (the static `stlink_open_usb_at()`) are missing.
6. **Library version at run time.** `version.h` only has compile time macros; a `stlink_lib_version()` lets applications check the loaded library.

[... cut ...]
### stlink_fs_posix.c (2)

| Function | Used outside the library | Used in the library | Recommendation |
|---|---|---|---|
| `search_for_chips` | – | chipid, stlink_fs_win32 | Internal |
| `stlink_exe_dir` | – | chipid, stlink_fs_win32 | Internal |

### stlink_threads_posix.c (2)

| Function | Used outside the library | Used in the library | Recommendation |
|---|---|---|---|
| `stlink_thread_create` | – | stlink_threads_win32, usb | Internal |
| `stlink_thread_join` | – | stlink_threads_win32, usb | Internal |

Nightwalker-87, 2026-10-08:
Update on resolved sub-tasks:
- logging (commit [#d464cd6](https://github.com/stlink-org/stlink/commit/d464cd69cc721c062e34ede035cf9e9acccff0fc)),
- API documentation (commits [#b7a27bd](https://github.com/stlink-org/stlink/commit/b7a27bda980d7ace513d6fcef70c6060bb82f9cb), [#6d4d4b3](https://github.com/stlink-org/stlink/commit/6d4d4b3cad7375ee52f01d427993128e618c1289)),
- hardware consolidation (commit [#387e8e2](https://github.com/stlink-org/stlink/commit/387e8e24b7733b147d33c53c1d7c2a489c823560)) and
- coding conventions (commits [#0e63cc9](https://github.com/stlink-org/stlink/commit/0e63cc903602837c696c1bfe8c6473aab5741602), [#6a47874](https://github.com/stlink-org/stlink/commit/6a478741cc7f2ba5ead429aeeaf7dec7541a308d)).

closed 2026-10-08 by Nightwalker-87