stlink-org/stlink #1475: WARN usb.c: skipping ST device : 0x483:0xdf11

MrNavaStar, 2025-08-27:
I am attempting to use some of these weird stm32f4 dev kits from [china](https://www.amazon.ca/Koarmy-Development-STM32F407VET6-Learning-Ethernet/dp/B0D49L4TL1?crid=7IVGWIQN08W8&dib=eyJ2IjoiMSJ9.a1rBKAcotrp4SyTEsdELdrY1peNrZPU-PoPZwzO8_hl1N30Vv3W6fvlcVa-o0d06sdpc-4dQyS7z80vGHmNbXQRGK8yd9YG5FXT4q8BGLkCuzVPkXZAsl1sRKHgvj2bW3t9Upzu320EvHt4VKMlO_Jq4jyPEzRePEk1nI2WI_44dXIbUK3oVcY2s63SrJJGlWK5Z-ETCGo-U4jSSf5InhD8RQQ_eV9f7l5HlpglaLNIDSzSBZ3ItU4wrU32SjJ88KwVOcz3bB_7XZAfvUivgpyn5D7ioSkqQdWxZy3fZ0NU.UouimPCWrBhIB4MVqLM9I2d1QWzyEK4ZoNEwpeq7gHM&dib_tag=se&keywords=stm32+ethernet+dev+board&qid=1756256934&sprefix=stm32+ethernet+dev+boar%2Caps%2C129&sr=8-6), and as one would expect im running into issues. the device doesn't seem to be properly recognized by `st-info --probe`, throwing: 

[image]

Im pretty sure a udev rule just needs to be added for the board? lmk if there is anything I can do to help. Cheers!

a-michelis, 2025-08-28:
Hello, [MrNavaStar](https://github.com/MrNavaStar)

Your device is reported to have 0xDF11 as PID. This means your device is in DFU mode. 
Can you share more about how you attempt connect the device to the computer?

MrNavaStar, 2025-08-28:
The board is just plugged in via micro usb

closed 2025-08-29 by a-michelis

MrNavaStar, 2025-08-29:
Aaahh okay, I appreciate it! Thanks so much